ansys/pyaedt #6119: Logger System Lacks Sufficient Detail - Request to Add Content from the Profile of the Design

z331225718, 2025-05-07:
### Description of the feature

After reviewing the API documentation, I found that only the output from the export_profile method meets the requirements. We propose integrating this profiling data directly into the logger system instead of requiring manual exports. The current workflow poses two critical issues:

Runtime conflict: Export operations cannot be performed concurrently while the application is actively running

User experience gap: Post-execution export contradicts user expectations, as operators frequently need to monitor real-time progress and estimate phase durations during ongoing processes.
By the way ,we are using it with non_graphical mode

Samuelopez-ansys, 2025-05-07:
Hi @z331225718 ,

Please add more information to this issue. 

PyAEDT has also this method: get_profile

https://aedt.docs.pyansys.com/version/stable/API/_autosummary/ansys.aedt.core.modules.solve_setup.SetupQ3D.get_profile.html#get_profile

This is updated dynamically. If you are using HFSS, when you run the analysis you can enable the non_blocking parameter, and then in a while loop you can call get_profile to get the simulation status.

z331225718, 2025-05-07:
I think get_profile return dict of ansys.aedt.core.modeler.cad.elements_3d.BinaryTree.The class of BinaryTreeNode description:
    """Manages an object's history structure.""" 
I think it is not what I want.
I want this info:

Solution setup: Setup1

  Task                  Real Time  CPU Time  Memory  Information
  ----                  ---------  --------  ------  -----------

Samuelopez-ansys, 2025-05-07:
Hi @z331225718 ,

Whit this simple code you can extract all profile data, then manipulate it as you want.

```
from ansys.aedt.core import Hfss

app = Hfss(version="2025.1", non_graphical=False, new_desktop=False)

app.analyze(blocking=False, setup="Advanced_31Ghz")

while app.are_there_simulations_running:
    profile = app.get_profile()
    keys = list(profile.keys())
    solution_keys = list(profile[keys[0]].children.keys())
    properties = profile[keys[0]].children[solution_keys[0]]
    properties.children["Initial Meshing Group"]
    
app.release_desktop(False, False)
```

You need to navigate through the dictionary, but remind that it is dynamic, because the profile ischaning during the simulation.

z331225718, 2025-05-08:
Thanks, it works when "non_graphical=False",if non_graphical,It will not enter the while loop until the analyze process is completed
？？

Samuelopez-ansys, 2025-05-08:
Hio @z331225718 ,

I am using a HFSS project in 2025R1 and it is working for me.

```
from ansys.aedt.core import Hfss

app = Hfss(version="2025.1",
           non_graphical=True,
           new_desktop=True, 
           project=r"C:\Users\slopez\Downloads\Potter_Horn_Antenna.aedt")

app.analyze(blocking=False, setup="Advanced_31Ghz")

while app.are_there_simulations_running:
    profile = app.get_profile()
    keys = list(profile.keys())
    solution_keys = list(profile[keys[0]].children.keys())
    properties = profile[keys[0]].children[solution_keys[0]]
    properties.children["Initial Meshing Group"]

app.release_desktop(False, False)
```