ansys/pyaedt #6120: export interactive 3d field plot

arifroni, 2025-05-07:
Hello, I have a field plot in Q3D. is there anyway to export this plot as an interactive plot?

arifroni, 2025-05-07:
> [@arifroni](https://github.com/arifroni) What do you mean by an interactive plot?
> 
> PyAEDT can export the field data to PyVista for instance using this method: plot_field
> 
> In the examples you have multiple use cases, for instance:
> 
> https://examples.aedt.docs.pyansys.com/version/dev/examples/high_frequency/multiphysics/mri.html

i mean a plot that i can rotate and see like i can in aedt. 
alternatively if i can export field values with coordinates i can use pyvista to make that plot. but i also didnt find any way to export the field data with coordinate. 

i can export aedtplt format file with export_field_plot command, but i dont know what to do with it.

Samuelopez-ansys, 2025-05-07:
Please review the example, the method plot_field is exactly doing it. It is exporting the .aedtplt file and loading it into PyVista.

You can see how it is implemented if you want something custom for your case. https://github.com/ansys/pyaedt/blob/main/src/ansys/aedt/core/visualization/post/post_common_3d.py#L2146

Samuelopez-ansys, 2025-05-07:
Even in this post you have some examples:

https://github.com/ansys/pyaedt/discussions/4459

Samuelopez-ansys, 2025-05-08:
@arifroni Please let us know if this issue can be closed

closed 2025-05-09 by Samuelopez-ansys