ansys/pyaedt #6163: Add remove_lock to Desktop class

jvela018, 2025-05-16:
### Description of the feature

Check and remove lock from aedt project if possible

### Steps for implementing the feature

Copy implementation from Hfss

gmalinve, 2025-05-17:
@jvela018, not sure if this is what you mean but we already have a method to remove the lock file: https://github.com/ansys/pyaedt/blob/main/src/ansys/aedt/core/generic/file_utils.py#L117

Samuelopez-ansys, 2025-05-19:
I think what  @jvela018 means is to add this option in the load_project method.

jvela018, 2025-05-19:
@gmalinve @Samuelopez-ansys , Indeed. It was the load_project method. I just changed the name of the issue. Sorry about the confusion

closed 2025-10-15 by jvela018