ansys/pyaedt #6187: Bug located in launch_aedt_in_lsf() inside desktop.py

kruglovIFX, 2025-05-22:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

The same `settings.custom_lsf_command` that was working on pyaedt 0.13.00 throws an error in 0.16.02.

After debugging, the error seems to appear after executing line 209 inside `desktop.py`, in the `launch_aedt_in_lsf()` function.

```
        p = subprocess.Popen(command, stdout=subprocess.PIPE, stderr=subprocess.PIPE)  # nosec
```

### Steps To Reproduce

Run this code on LSF. Works on 0.13.00, throws an error on 0.16.02

```
""" Required LSF settings for the interactive sessions """
grpcsrv = ''                        # Keep this setting (required for LSF)
settings.use_lsf_scheduler = True   # set True if using LSF, False to run locally
settings.custom_lsf_command = 'bsub <company's settings> ansysedt'
 
desktop = Desktop()

```

Error:

[image]

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

aiohappyeyeballs==2.4.4
aiohttp==3.11.11
aiosignal==1.3.2
alabaster==1.0.0
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc4
ansys-tools-visualization-interface==0.6.2
anyio==4.9.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
astroid==3.3.10
asttokens==3.0.0
async-lru==2.0.5
async-timeout==5.0.1
asyncssh==2.21.0
atomicwrites==1.4.1
attrs==24.2.0
autopep8==2.0.4
babel==2.17.0
backports.tarfile==1.2.1
bcrypt==4.3.0
beautifulsoup4==4.13.4
binaryornot==0.4.4
black==25.1.0
bleach==6.2.0
blosc2==2.7.1
certifi==2024.12.14
cffi==1.17.1
chardet==5.2.0
charset-normalizer==3.4.0
click==8.2.1
cloudpickle==3.1.1
clr_loader==0.2.7.post0
comm==0.2.2
contourpy==1.3.1
cookiecutter==2.6.0
cryptography==45.0.2
cycler==0.12.1
debugpy==1.8.14
decorator==5.2.1
defusedxml==0.7.1
Deprecated==1.2.18
diff-match-patch==20241021
dill==0.4.0
distro==1.9.0
docstring-to-markdown==0.17
docutils==0.21.2
dotnetcore2==3.1.23
et_xmlfile==2.0.0
exceptiongroup==1.3.0
executing==2.2.0
fast_simplification==0.1.9
fastjsonschema==2.21.1
flake8==7.1.2
fonttools==4.55.3
fpdf2==2.8.2
fqdn==1.5.1
frozenlist==1.5.0
geopandas==1.0.1
grpcio==1.70.0
h11==0.16.0
httpcore==1.0.9
httpx==0.28.1
idna==3.10
imageio==2.36.1
imagesize==1.4.1
importlib_metadata==8.7.0
inflection==0.5.1
intervaltree==3.1.0
ipykernel==6.29.5
ipython==8.36.0
ipywidgets==8.1.7
isoduration==20.11.0
isort==6.0.1
jaraco.classes==3.4.0
jaraco.context==6.0.1
jaraco.functools==4.1.0
jedi==0.19.2
jeepney==0.9.0
jellyfish==1.2.0
Jinja2==3.1.6
joblib==1.4.2
json5==0.12.0
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
jupyter==1.1.1
jupyter-console==6.6.3
jupyter-events==0.12.0
jupyter-lsp==2.2.5
jupyter_client==8.6.3
jupyter_core==5.7.2
jupyter_server==2.16.0
jupyter_server_terminals==0.5.3
jupyterlab==4.4.2
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.3
jupyterlab_widgets==3.0.15
keyring==25.6.0
kiwisolver==1.4.7
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.9.4
matplotlib-inline==0.1.7
mccabe==0.7.0
mdurl==0.1.2
meshio==5.3.5
mistune==3.1.3
more-itertools==10.5.0
msgpack==1.1.0
multidict==6.1.0
mypy_extensions==1.1.0
nbclient==0.10.2
nbconvert==7.16.6
nbformat==5.10.4
ndindex==1.9.2
nest-asyncio==1.6.0
networkx==3.4.2
notebook==7.4.2
notebook_shim==0.2.4
numexpr==2.10.2
numpy==2.2.0
numpydoc==1.8.0
openpyxl==3.1.5
osmnx==2.0.0
overrides==7.7.0
packaging==24.2
pandas==2.2.3
pandocfilters==1.5.1
parso==0.8.4
pathspec==0.12.1
pexpect==4.9.0
pickleshare==0.7.5
pillow==11.0.0
platformdirs==4.3.6
pluggy==1.6.0
plumbum==1.9.0
pooch==1.8.2
prometheus_client==0.22.0
prompt_toolkit==3.0.51
propcache==0.2.1
psutil==6.1.1
ptyprocess==0.7.0
pure_eval==0.2.3
py-cpuinfo==9.0.0
pyaedt==0.16.2
pycodestyle==2.12.1
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pydocstyle==6.3.0
pyedb==0.34.3
pyflakes==3.2.0
PyGithub==2.6.1
Pygments==2.18.0
PyJWT==2.10.1
pylint==3.3.7
pylint-venv==3.0.4
pyls-spyder==0.4.0
PyNaCl==1.5.0
pyogrio==0.10.0
pyparsing==3.2.0
pyproj==3.7.0
PyQt5==5.15.11
PyQt5-Qt5==5.15.16
PyQt5_sip==12.17.0
PyQtWebEngine==5.15.7
PyQtWebEngine-Qt5==5.15.16
python-dateutil==2.9.0.post0
python-json-logger==3.3.0
python-lsp-black==2.0.0
python-lsp-jsonrpc==1.1.2
python-lsp-server==1.12.2
python-slugify==8.0.4
pytoolconfig==1.3.1
pytz==2024.2
pyuca==1.2
pyvista==0.44.2
pyxdg==0.28
PyYAML==6.0.2
pyzmq==26.4.0
QDarkStyle==3.2.3
qstylizer==0.2.4
QtAwesome==1.4.0
qtconsole==5.6.1
QtPy==2.4.3
referencing==0.35.1
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rich==13.9.4
rope==1.13.0
rpds-py==0.22.3
rpyc==6.0.1
Rtree==1.3.0
scikit-learn==1.6.0
scikit-rf==1.5.0
scipy==1.14.1
scooby==0.10.0
SecretStorage==3.3.3
Send2Trash==1.8.3
shapely==2.0.6
six==1.17.0
sniffio==1.3.1
snowballstemmer==3.0.1
sortedcontainers==2.4.0
soupsieve==2.7
Sphinx==8.1.3
sphinxcontrib-applehelp==2.0.0
sphinxcontrib-devhelp==2.0.0
sphinxcontrib-htmlhelp==2.1.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==2.0.0
sphinxcontrib-serializinghtml==2.0.0
spyder==6.0.6
spyder-kernels==3.0.4
SRTM.py==0.3.7
stack-data==0.6.3
superqt==0.7.3
tables==3.10.1
tabulate==0.9.0
terminado==0.18.1
text-unidecode==1.3
textdistance==4.6.3
threadpoolctl==3.5.0
three-merge==0.1.1
tinycss2==1.4.0
toml==0.10.2
tomli==2.2.1
tomli_w==1.1.0
tomlkit==0.13.2
tornado==6.5
traitlets==5.14.3
trame==3.7.1
trame-client==3.5.0
trame-server==3.2.3
trame-vtk==2.8.12
trame-vuetify==2.7.2
types-python-dateutil==2.9.0.20250516
typing_extensions==4.12.2
tzdata==2024.2
ujson==5.10.0
uri-template==1.3.0
urllib3==2.2.3
utm==0.7.0
vtk==9.3.1
watchdog==6.0.0
wcwidth==0.2.13
webcolors==24.11.1
webencodings==0.5.1
websocket-client==1.8.0
websockets==14.1
whatthepatch==1.0.7
widgetsnbextension==4.0.14
wrapt==1.17.2
wslink==2.2.1
wurlitzer==3.1.1
yapf==0.43.0
yarl==1.18.3
zipp==3.21.0

SMoraisAnsys, 2025-05-26:
Thanks for reporting that @kruglovIFX, I'll check what changed and could explain the failure.

Devin-Crawford, 2025-05-27:
The command ``settings.custom_lsf_command.split(" ")`` split the resource
request passed to ``-R``. This resulted in an error when the resulting list was passed to 
``subprocess.Popen()``.

``command = shlex.split(settings.custom_lsf_command)`` fixes this.

kruglovIFX, 2025-06-03:
@Devin-Crawford @SMoraisAnsys just tried with 0.17.00 - the issue persists