ansys/pyaedt #6230: Iron Python conversion to PyAEDT: oModule.ToggleNet & oModule.RenameBoundary

razvanmoisa1331, 2025-06-04:
I'm trying to convert from an iron script to PyAEDT, but I couldn't find the equivalent functions for these two functions:
oModule.ToggleNet("Name", "FloatingNet")
oModule.RenameBoundary("OldName", "NewName")

I found a work around that works, but it is time consuming:
nets=app.nets
for net_name in nets:
 sink_list = app.net_sinks(net_name)
 if not sink_list:
 obj_list = app.objects_from_nets(net_name)
 app.assign_net(obj_list[net_name], net_type="Floating")
 next(b for b in app. boundaries if b.name == net_name).delete()
 elif len(sink_list) == 1:
 position = list_name.index(sink_list[0])
 b = next(b for b in b in app.boundaries if b.name == net_name);b.name = list_net_name[position];b.update()

Has anyone encountered this problem so far?

Samuelopez-ansys, 2025-06-05:
Hi @razvanmoisa1331 ,

What do you exactly need?

You can get the available sink objects with this simple call:

app.boundaries_by_type["Sink"]

sink1 = app.boundaries_by_type["Sink"][0]

sink1.name = "my_new_sink"

The toggle is still not implemented, If you describe what you need, we could consider adding it.

Samuelopez-ansys, 2025-06-06:
Hi @razvanmoisa1331,

Could you please share a dummy project to reproduce the workflow? And also the Ironpython code would help.

Regards

razvanmoisa1331, 2025-06-06:
Hi @Samuelopez-ansys, 

I put the pyAEDT code below. It contains, commented, how the equivalent function looks like in ironPython. This is a very simplified model, that's why it works fast, but when you're talking about 30-40 nets, the time in which all the steps are done increases exponentially.

###################### CODE ########################### 
import ansys.aedt.core
from pyaedt import Circuit, Q3d, Desktop

desktop = Desktop(version = "2024.2", non_graphical=False,desktop=True,close_on_exit = False)
project_name = r"C:\............\test.aedt"
app = Q3d(project=project_name, design="test")
app.autosave_disable()

id = []
points_polyline = [["54.45mm","5.15mm","2.274mm"],["68.15mm","5.15mm","2.274mm"],["68.15mm","12.43mm","2.274mm"],["54.45mm","12.43mm","2.274mm"]]
app.modeler.create_polyline(points = points_polyline,name = "Busbar_HVp_1", segment_type=None, cover_surface=True, close_surface=True,material = "copper", non_model=False)
app.modeler.thicken_sheet(assignment = "Busbar_HVp_1", thickness = "4mm", both_sides = False)
id.append("Busbar_HVp_1")

points_polyline = [["80.15mm","5.15mm","2.274mm"],["94.15mm","5.15mm","2.274mm"],["94.15mm","12.43mm","2.274mm"],["80.15mm","12.43mm","2.274mm"]]
app.modeler.create_polyline(points = points_polyline,name = "Busbar_HVp_2", segment_type=None, cover_surface=True, close_surface=True,material = "copper", non_model=False)
app.modeler.thicken_sheet(assignment = "Busbar_HVp_2", thickness = "4mm", both_sides = False)
id.append("Busbar_HVp_2")

points_polyline = [["55.15mm","4.15mm","6.274mm"],["105.55mm","4.15mm","6.274mm"],["105.55mm","37.15mm","6.274mm"],["55.15mm","37.15mm","6.274mm"]]
app.modeler.create_polyline(points = points_polyline,name = "Busbar_HVp_3", segment_type=None, cover_surface=True, close_surface=True,material = "copper", non_model=False)
app.modeler.thicken_sheet(assignment = "Busbar_HVp_3", thickness = "0.8mm", both_sides = False)
id.append("Busbar_HVp_3")

app.modeler.unite(id)

id = []
points_polyline = [["94.15mm","47.15mm","2.274mm"],["80.15mm","47.15mm","2.274mm"],["80.15mm","39.82mm","2.274mm"],["94.15mm","39.82mm","2.274mm"]]
app.modeler.create_polyline(points = points_polyline,name = "Busbar_HVm_1", segment_type=None, cover_surface=True, close_surface=True,material = "copper", non_model=False)
app.modeler.thicken_sheet(assignment = "Busbar_HVm_1", thickness = "6.1mm", both_sides = False)
id.append("Busbar_HVm_1")

points_polyline = [["68.15mm","47.15mm","2.274mm"],["54.15mm","47.15mm","2.274mm"],["54.15mm","39.82mm","2.274mm"],["68.15mm","39.82mm","2.274mm"]]
app.modeler.create_polyline(points = points_polyline,name = "Busbar_HVm_2", segment_type=None, cover_surface=True, close_surface=True,material = "copper", non_model=False)
app.modeler.thicken_sheet(assignment = "Busbar_HVm_2", thickness = "6.1mm", both_sides = False)
id.append("Busbar_HVm_2")

points_polyline = [["54.15mm","47.15mm","8.374mm"],["105.55mm","47.15mm","8.374mm"],["105.55mm","4.15mm","8.374mm"],["54.15mm","4.15mm","8.374mm"]]
app.modeler.create_polyline(points = points_polyline,name = "Busbar_HVm_3", segment_type=None, cover_surface=True, close_surface=True,material = "copper", non_model=False)
app.modeler.thicken_sheet(assignment = "Busbar_HVm_3", thickness = "0.8mm", both_sides = False)
id.append("Busbar_HVm_3")

app.modeler.unite(id)

id=app.modeler.get_faceid_from_position([54.45, 8.79, 4.274])
app.source(assignment = id, name = "HVp_source",terminal_type = "current")
id=app.modeler.get_faceid_from_position([105.55, 20.65, 6.674])
app.sink(assignment = id, name = "HVp",terminal_type = "current")

app.auto_identify_nets()

#This piece of code replaces the toggleNet part of ironpython
net_name='Busbar_HVm_1'
obj_list = app.objects_from_nets(net_name)
app.assign_net(obj_list[net_name], net_type="Floating")
next(b for b in app.boundaries if b.name == net_name).delete()
''' iron python code
oModule = oDesign.GetModule("BoundarySetup")
oModule.ToggleNet("Busbar_HVm_1", "FloatingNet")
'''

#This code replaces the Net rename part of ironpython
net_name = 'Busbar_HVp_1'
b = next(b for b in app.boundaries if b.name == net_name);b.name = "net_HVp";b.update()
''' iron python code
oModule = oDesign.GetModule("BoundarySetup")
oModule.RenameBoundary("Busbar_HVp_1", "net_HVp")
'''
app.release_desktop()

Samuelopez-ansys, 2025-06-06:
The rename as I said, was already possible, you just need to get the specific net, like 

p = app.boundaries[0]
p.name = "new_name"

PR #6237 (2025-06-06): FEAT: Toggle net type in Q3D
## Description
This new method toggles the net type: Signal, Float and Ground. If the net contains sources and sinks, they are automatically removed, because Float and Ground can not have sources or sinks. 

## Issue linked
Close #6230 

## Checklist
- [ ] I have tested my changes locally.
- [ ] I have added necessary documentation or updated existing documentation.
- [ ] I have followed the coding style guidelines of this project.
- [ ] I have added appropriate tests (unit, integration, system).
- [ ] I have reviewed my changes before submitting this pull request. [... cut]