ansys/pyaedt #6247: Bug located in insert_layout_component

hui-zhou-a, 2025-06-09:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

Fail to import layout coordinate systems

### Steps To Reproduce

`
 hfss.modeler.insert_layout_component(
            input_file=self.file_path,
            coordinate_system=self.target_coordinate_system,
            reference_coordinate_system=self.reference_coordinate_system,
            layout_coordinate_systems="CABLE1",
            name="pcb",
        )
`

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

all

Samuelopez-ansys, 2025-06-16:
Hi @hui-zhou-a ,

Last week we implemented the HFSS 3D Layout coordinate system class. Please could you try?

hui-zhou-a, 2025-06-17:
@Samuelopez-ansys that method handles coordinate system of the opened project. It doesn't help in this PR.