ansys/pyaedt #6395: Example Code to Export Touchstone File from HFSS Simulation

girdharic, 2025-07-14:
I have created and successfully simulated a model in HFSS. After completing the simulation, I would like to export the S-parameter data as a Touchstone (.s2p) file and save it to a specific folder on my system.

Could you kindly guide me on how to write a script to perform this task? 
Could you provide an example code that demonstrates how to automate the export of the Touchstone file after simulation.

Samuelopez-ansys, 2025-07-14:
Hi, 

Please share more information about the issue, because you probably missed passing the variations. I prefer the auto export on completion because you will not dependent on the variation.

girdharic, 2025-07-14:
PyAEDT INFO: Exporting Touchstone D:/Projects/ANSYS HFSS/BPF/example.s2p
PyAEDT ERROR: **************************************************************
PyAEDT ERROR:   File "C:\Program Files\AnsysEM\v242\Win64\commonfiles\CPython\3_10\winx64\Release\python\lib\runpy.py", line 196, in _run_module_as_main
PyAEDT ERROR:     return _run_code(code, main_globals, None,
PyAEDT ERROR:   File "C:\Program Files\AnsysEM\v242\Win64\commonfiles\CPython\3_10\winx64\Release\python\lib\runpy.py", line 86, in _run_code
PyAEDT ERROR:     exec(code, run_globals)
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ipykernel_launcher.py", line 18, in <module>
PyAEDT ERROR:     app.launch_new_instance()
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\traitlets\config\application.py", line 1075, in launch_instance
PyAEDT ERROR:     app.start()
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ipykernel\kernelapp.py", line 739, in start
PyAEDT ERROR:     self.io_loop.start()
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\tornado\platform\asyncio.py", line 205, in start
PyAEDT ERROR:     self.asyncio_loop.run_forever()
PyAEDT ERROR:   File "C:\Program Files\AnsysEM\v242\Win64\commonfiles\CPython\3_10\winx64\Release\python\lib\asyncio\base_events.py", line 603, in run_forever
PyAEDT ERROR:     self._run_once()
PyAEDT ERROR:   File "C:\Program Files\AnsysEM\v242\Win64\commonfiles\CPython\3_10\winx64\Release\python\lib\asyncio\base_events.py", line 1909, in _run_once
PyAEDT ERROR:     handle._run()
PyAEDT ERROR:   File "C:\Program Files\AnsysEM\v242\Win64\commonfiles\CPython\3_10\winx64\Release\python\lib\asyncio\events.py", line 80, in _run
PyAEDT ERROR:     self._context.run(self._callback, *self._args)
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ipykernel\kernelbase.py", line 545, in dispatch_queue
PyAEDT ERROR:     await self.process_one()
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ipykernel\kernelbase.py", line 534, in process_one
PyAEDT ERROR:     await dispatch(*args)
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ipykernel\kernelbase.py", line 437, in dispatch_shell
PyAEDT ERROR:     await result
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ipykernel\ipkernel.py", line 362, in execute_request
PyAEDT ERROR:     await super().execute_request(stream, ident, parent)
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ipykernel\kernelbase.py", line 778, in execute_request
PyAEDT ERROR:     reply_content = await reply_content
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ipykernel\ipkernel.py", line 449, in do_execute
PyAEDT ERROR:     res = shell.run_cell(
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ipykernel\zmqshell.py", line 549, in run_cell
PyAEDT ERROR:     return super().run_cell(*args, **kwargs)
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Local\Temp\ipykernel_3920\2130240239.py", line 5, in <module>
PyAEDT ERROR:     hfss.export_touchstone(
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ansys\aedt\core\application\analysis_hf.py", line 380, in export_touchstone
PyAEDT ERROR:     return self._app._export_touchstone(
PyAEDT ERROR:   File "C:\Users\JBNU11\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ansys\aedt\core\application\analysis.py", line 2031, in _export_touchstone
PyAEDT ERROR:     self.osolution.ExportNetworkData(
PyAEDT ERROR: AEDT API Error on _export_touchstone
PyAEDT ERROR: Last Electronics Desktop Message - [error] script macro error: solution data is not available.  it was either not solved or was made invalid by subsequent design changes. (04:51:58 pm  jul 14, 2025)

PyAEDT ERROR: Method arguments: 
PyAEDT ERROR:     setup_name = Setup1 
PyAEDT ERROR:     sweep_name = Sweep1 
PyAEDT ERROR:     file_name = D:\Projects\ANSYS HFSS\BPF\example.s2p 
PyAEDT ERROR:     renormalization = True 
PyAEDT ERROR:     impedance = 50 
PyAEDT ERROR: **************************************************************
False

girdharic, 2025-07-14:
Is it possible to specify desired file name and path to save file while using auto export in completion?

Samuelopez-ansys, 2025-07-14:
@girdharic Yes, the second argument is the output_dir

You can not specify the file name because HFSS will change the name depending on the variation, setup and sweep. 

Regarding your error, I would need a dummy project to reproduce the issue, because very likely, you need to add the variation.