ansys/pyaedt #6488: Incorrect information in breaking change note

hui-zhou-a, 2025-08-02:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

Following the guidelines, fails to install .NET on Linux

[image]

By MS official information, .NET 9.0 is recommended.
[image]

### Steps To Reproduce

NA

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

all

SMoraisAnsys, 2025-08-02:
You havent added the Microsoft registre as explained in the note. Could you give it a try.
I gollowed the documentation and succeded to instal donet without issues.

hui-zhou-a, 2025-08-04:
I successfullu installed the required packages, but when running PyEDB I get below error.
[image]