ansys/pyaedt #6648: Maxwell extension

climaxburden, 2025-09-15:
I followed the help of the webpage, but the extension never popped up. I set the port number and version number as required.
How to use this extension? Is there a detailed case tutorial? I have tried it on the webpage. I just want to export the Mag_B data of the specified node set

extension：
https://aedt.docs.pyansys.com/version/stable/User_guide/pyaedt_extensions_doc/commandline.html

Command：
C:\Users\lenovo>SET PYAEDT_SCRIPT_PORT=50051

C:\Users\lenovo>SET PYAEDT_SCRIPT_VERSION=2025.1

C:\Users\lenovo>python.exe path/to/E:/AnacondaEnvs/JQXX/Lib/site-packages/ansys/aedt/core/extensions/project/import_nastran.py

C:\Users\lenovo>

climaxburden, 2025-09-18:
> Hi [@climaxburden](https://github.com/climaxburden),  Hi ，
> 
> You can open the extension manager directly on the User Interface. If you installed pyaedt using the interface as shown below, you'll automatically have the extension manager available in the GUI under the Automation tab您可以直接在用户界面上打开扩展管理器。如果您像下面这样通过界面安装了 pyaedt，那么扩展管理器将自动在 GUI 的自动化选项卡下可用。
> 
> <img alt="Image" width="2000" height="1274" src="https://private-user-images.githubusercontent.com/39401808/490510957-a4e769b6-1e75-4dc0-8355-8845a39a1374.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTgxNjUzNjksIm5iZiI6MTc1ODE2NTA2OSwicGF0aCI6Ii8zOTQwMTgwOC80OTA1MTA5NTctYTRlNzY5YjYtMWU3NS00ZGMwLTgzNTUtODg0NWEzOWExMzc0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTE4VDAzMTEwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQwYTE4Y2ZkZjk1ZWRlNjkzMWEwMmQ0M2MzNmQ5NTYxYTIzMzAwNjc2MzdmYzA2ODc1OWU2YzY0YjBkYzBlYjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.F-XN527WSMI-1LLCnu3Vdm-C0MiVGY7D8PKUQisL9qg">
> I hope this helps.希望这能帮到您。
> 
> /jvela018

Firstly, thank you for your reply. What I'm actually seeking to understand is whether it's possible to invoke the functionality within the extension via Python, rather than merely opening the extension interface through Python or the terminal and then manually clicking on the required features. What I'm seeking to ascertain is whether the extension's functions have corresponding Python interface functions or similar mechanisms. This is because I wish to automate the process rather than rely on manual clicks. For instance, regarding the Maxwell2D extension's ‘Fields distribution’ feature, could this extension be automated via a Python script to import node sets and export MagB data, rather than performing these actions manually?

jvela018, 2025-09-19:
@climaxburden ,

One last thing. Now that I'm re-reading what you asked for - you can run the script directly using the command 

`python field_distribution.py`

The only thing, is that you must have an AEDT instance with a valid design to be able to benefit from the extension.

An alternative would be to create another script and use the backend of the extension directly for example:

import ansys.aedt.core
import os
from ansys.aedt.core.extensions.project.field_distribution import main
file_path = "my_file.stl"
m3d= ansys.aedt.core.Maxwell3D()
# Specify the AEDT session to connect
os.environ["PYAEDT_SCRIPT_PORT"] = str(m3d.desktop_class.port)
os.environ["PYAEDT_SCRIPT_VERSION"] = m3d.desktop_class.aedt_version_id
# Launch extension
main({"file_path": file_path, "lightweight": True, "decimate": 0.0, "planar": True, "is_test": False})

Please, let me know if this is helpful.

/jvela018